OAID/Tengine #1247: 求助：在khadas VIM3上编译tengine-lite出错

zhanggd18, 2021-12-31:
环境：板子：Fenix 1.0.7 Ubuntu 20.04.2 LTS Linux 4.9.241，aml-npu/focal,now 6.4.6.2.5.3.2 arm64

1.编译之前板子上/usr/lib下有：libCLC.so  libVSC.so

2.编译过程中：
  /usr/bin/ld: cannot find -lCLC
  /usr/bin/ld: cannot find -lVSC
  collect2: error: ld returned 1 exit status
  make[2]: *** [source/CMakeFiles/tengine-lite.dir/build.make:6980: source/libtengine-lite.so] Error 1
  make[1]: *** [CMakeFiles/Makefile2:316: source/CMakeFiles/tengine-lite.dir/all] Error 2
  make: *** [Makefile:160: all] Error 2

同时下发/usr/lib下：libCLC.so  libVSC.so都不见了

tianyiqwq, 2022-01-15:
大佬解决了吗，我也遇到这个问题了

zhanggd18, 2022-01-15:
> 大佬解决了吗，我也遇到这个问题了

没有解决...

tianyiqwq, 2022-01-17:
> > 大佬解决了吗，我也遇到这个问题了
> 
> 没有解决...

解决了，我问了群里的大佬，是TIM-VX版本太新了，进到tim-vx目录里 git checkout 68b5acb，再来一遍成功了

zhanggd18, 2022-01-22:
> 

十分感谢