OAID/Tengine #1253: rv64 make

YanxingLiu, 2022-01-12:


YanxingLiu, 2022-01-12:
I want to build Tengine for allwinnertech D1 SBC. And I follow the instructions in this link https://zhuanlan.zhihu.com/p/368640131. But I got an error of 
riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-mfp16'. 
[image]

I have tried to delete flag -mfp16 in toolchains/rv64-c906.toolchain.cmake, but it still did not work.

BUG1989, 2022-01-18:
@nihui Please help us......

YanxingLiu, 2022-02-01:
I fixed this issue by changing compling toolchain. 
the toolchain URL is: https://occ.t-head.cn/community/download?id=3890532453097742336

closed 2022-02-01 by YanxingLiu