OAID/Tengine #1257: Source building for NXP i.MX 8M Plus

fengwang, 2022-01-17:
## Environment:

$ uname -a
Linux imx8mpevk 5.4.70-2.3.0+g4f2631b022d8 #1 SMP PREEMPT Mon Mar 8 15:39:16 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/aarch64-poky-linux/9.2.0/lto-wrapper
Target: aarch64-poky-linux
Configured with: ../../../../../../work-shared/gcc-9.2.0-r0/gcc-9.2.0/configure --build=x86_64-linux --host=aarch64-poky-linux --target=aarch64-poky-linux --prefix=/usr --exec_prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/libexec --datadir=/usr/share --sysconfdir=/etc --sharedstatedir=/com --localstatedir=/var --libdir=/usr/lib --includedir=/usr/include --oldincludedir=/usr/include --infodir=/usr/share/info --mandir=/usr/share/man --disable-silent-rules --disable-dependency-tracking --with-libtool-sysroot=/ --with-gnu-ld --enable-shared --enable-languages=c,c++ --enable-threads=posix --enable-multilib --enable-default-pie --enable-c99 --enable-long-long --enable-symvers=gnu --enable-libstdcxx-pch --program-prefix=aarch64-poky-linux- --without-local-prefix --enable-lto --disable-libssp --enable-libitm --disable-bootstrap --disable-libmudflap --with-system-zlib --with-linker-hash-style=gnu --enable-linker-build-id --with-ppl=no --with-cloog=no --enable-checking=release --enable-cheaders=c_global --without-isl --with-build-sysroot=/ --disable-static --enable-nls --with-glibc-version=2.28 --enable-initfini-array --enable-__cxa_atexit
Thread model: posix
gcc version 9.2.0 (GCC)

## Reproduction

####  clone/download source
> git clone https://github.com/VeriSilicon/TIM-VX.git 
> git clone https://github.com/OAID/Tengine.git tengine-lite 
> wget -c https://github.com/VeriSilicon/TIM-VX/releases/download/v1.1.28/aarch64_S905D3_D312513_A294074_R311680_T312233_O312045.tgz && tar xvf aarch64_S905D3_D312513_A294074_R311680_T312233_O312045.tgz && mv aarch64_S905D3_D312513_A294074_R311680_T312233_O312045 prebuild-sdk-s905d3

#### prepare files
> cd tengine-lite
> mkdir -p ./source/device/tim-vx/ && /bin/cp -rf ../TIM-VX/include ./source/device/tim-vx/ && /bin/cp -rf ../TIM-VX/src ./source/device/tim-vx/ 
> mkdir -p ./3rdparty/tim-vx/include && mkdir -p ./3rdparty/tim-vx/lib/aarch64 && /bin/cp -rf ../prebuild-sdk-s905d3/include/*  ./3rdparty/tim-vx/include/

#### compile

> mkdir -p build && cd ./build && cmake -DTENGINE_ENABLE_TIM_VX=ON .. && make -j4 VERBOSE=1 && make install

## Produced Error Message:

```
[  5%] Building C object source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/conv1d_ovxlib_cpu.c.o
cd /home/root/feng/Tengine_Second/tengine-lite/build/source/device/tim-vx && /usr/bin/cc -DOVXLIB_API="__attribute__((visibility(\"default\")))" -I/home/root/feng/Tengine_Second/tengine-lite/3rdparty/tim-vx/include -I/home/root/feng/Tengine_Second/tengine-lite/3rdparty/tim-vx/include/CL -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/include -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/include/tim/vx -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx/internal/include -O3 -DNDEBUG -fPIC -std=gnu99 -MD -MT source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/conv1d_ovxlib_cpu.c.o -MF CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/conv1d_ovxlib_cpu.c.o.d -o CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/conv1d_ovxlib_cpu.c.o -c /home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx/internal/src/kernel/cpu/conv1d_ovxlib_cpu.c
[  5%] Building C object source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_conv2d_cpu.c.o
cd /home/root/feng/Tengine_Second/tengine-lite/build/source/device/tim-vx && /usr/bin/cc -DOVXLIB_API="__attribute__((visibility(\"default\")))" -I/home/root/feng/Tengine_Second/tengine-lite/3rdparty/tim-vx/include -I/home/root/feng/Tengine_Second/tengine-lite/3rdparty/tim-vx/include/CL -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/include -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/include/tim/vx -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx/internal/include -O3 -DNDEBUG -fPIC -std=gnu99 -MD -MT source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_conv2d_cpu.c.o -MF CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_conv2d_cpu.c.o.d -o CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_conv2d_cpu.c.o -c /home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx/internal/src/kernel/cpu/cpu_backend_conv2d_cpu.c
/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx/internal/src/kernel/cpu/cpu_backend_conv2d_cpu.c:38:10: fatal error: cpu_backend/npuref_interface.h: No such file or directory
   38 | #include "cpu_backend/npuref_interface.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/build.make:1378: source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_conv2d_cpu.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[  5%] Building C object source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_deconv2d_cpu.c.o
cd /home/root/feng/Tengine_Second/tengine-lite/build/source/device/tim-vx && /usr/bin/cc -DOVXLIB_API="__attribute__((visibility(\"default\")))" -I/home/root/feng/Tengine_Second/tengine-lite/3rdparty/tim-vx/include -I/home/root/feng/Tengine_Second/tengine-lite/3rdparty/tim-vx/include/CL -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/include -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/include/tim/vx -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx -I/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx/internal/include -O3 -DNDEBUG -fPIC -std=gnu99 -MD -MT source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_deconv2d_cpu.c.o -MF CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_deconv2d_cpu.c.o.d -o CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_deconv2d_cpu.c.o -c /home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx/internal/src/kernel/cpu/cpu_backend_deconv2d_cpu.c
/home/root/feng/Tengine_Second/tengine-lite/source/device/tim-vx/src/tim/vx/internal/src/kernel/cpu/cpu_backend_deconv2d_cpu.c:38:10: fatal error: cpu_backend/npuref_interface.h: No such file or directory
   38 | #include "cpu_backend/npuref_interface.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/build.make:1392: source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/src/tim/vx/internal/src/kernel/cpu/cpu_backend_deconv2d_cpu.c.o] Error 1
make[2]: Leaving directory '/run/media/mmcblk2p2/home/root/feng/Tengine_Second/tengine-lite/build'
make[1]: *** [CMakeFiles/Makefile2:414: source/device/tim-vx/CMakeFiles/tim_ovx_internal.dir/all] Error 2
make[1]: Leaving directory '/run/media/mmcblk2p2/home/root/feng/Tengine_Second/tengine-lite/build'
make: *** [Makefile:146: all] Error 2
```

l1nfp, 2022-01-18:
I got the same linking error, also the same symbols are not found. Did you fix it ? @fengwang

fengwang, 2022-01-18:
> I got the same linking error, also the same symbols are not found. Did you fix it ? @fengwang

Not yet

fengwang, 2022-01-18:
@l1nfp Compiles if you revert TIM-VX back to Aug. 2021. Try `git checkout ff22b3e34c85e7a0737f4cac0600f037ccd5db03`  at the TIM-VX repo before copying.

l1nfp, 2022-01-18:
awesome,that works

BUG1989, 2022-01-18:
@fengwang Thanks for your help.

closed 2022-01-18 by fengwang

gdh1995, 2022-01-19:
Commit https://github.com/OAID/Tengine/commit/00f31fcce72f328c44fc9a831a8e6b0e6d4d1f42 suggests use https://github.com/VeriSilicon/TIM-VX/commit/68b5acbe7c90bceae073d4bfca60ffa92dff0fc4 , and it works.