OAID/Tengine #1266: bug report

TingfengTang, 2022-01-20:
Hi ,the micro defined below will cause substitution error:
`#define SET_LOG_PREFIX(prefix)                        \
    do                                                \
    {                                                 \
        struct logger* logger = get_default_logger(); \
        logger->prefix = prefix;                      \
    } while (0)
`
the code SET_LOG_PREFIX("myLogPrefix") will be expand to:
`
    do                                                \
    {                                                 \
        struct logger* logger = get_default_logger(); \
        logger->**"myLogPrefix"**= "myLogPrefix";                      \
    } while (0)
`
which is obviously wrong.

RudyFoo, 2022-01-20:
I fix it via https://github.com/OAID/Tengine/pull/1268

TingfengTang, 2022-01-21:
The fixed code may be covered by others,the latest code on github is still wrong.

closed 2022-02-02 by BUG1989