OAID/Tengine #1269: failed to cmake in rk3399

berylyellow, 2022-01-26:
输入下面指令cmake的时候：
cmake  -DCMAKE_TOOLCHAIN_FILE=../toolchains/aarch64-linux-gnu.toolchain.cmake                    
            -DTENGINE_ENABLE_ACL=ON ..
报错：
-- Using CMake tool chain file /home/firefly/project/Tengine/toolchains/aarch64-linux-gnu.toolchain.cmake
You have called ADD_LIBRARY for library tengine-lite-static without any source files. This typically indicates a problem with your CMakeLists.txt file
You have called ADD_LIBRARY for library tengine-lite without any source files. This typically indicates a problem with your CMakeLists.txt file
CMake Error at source/CMakeLists.txt:264 (TARGET_LINK_DIRECTORIES):
  Unknown CMake command "TARGET_LINK_DIRECTORIES".

定位发现，source/CMakeLists.txt中有个TEngine_NEXT_LIB_SRCS 没有定义：
ADD_LIBRARY (${TENGINE_LITE_NAME}-static STATIC ${TEngine_NEXT_LIB_SRCS})
ADD_LIBRARY (${TENGINE_LITE_NAME}        SHARED ${TEngine_NEXT_LIB_SRCS})

请问TEngine_NEXT_LIB_SRCS是指代哪些源文件呢？

liujiaxing7, 2022-04-19:
请问你解决了吗，我也是这个问题

liujiaxing7, 2022-04-20:
cmake的版本号问题