OAID/Tengine #1272: 在zcu102上运行出现core dump

Gitkingly, 2022-01-30:
错误信息是double free detected：
[image]

wei8171023, 2022-09-26:
遇到了相似的问题：
*** Error in `./tm_classification_opendla': double free or corruption (!prev): 0x0000000001b566d0 ***
Aborted

wei8171023, 2022-09-26:
> hw: zcu102+nvdla os: ubuntu20.04 Tengine: 1.5-dev
> 
> 错误信息是double free detected： [image]

我和你的推理输出结果事一样的，但是和官方README里执行结果并不一样，好奇是哪里不一致？！

wei8171023, 2022-10-25:
刚才测试了一下，分别用CPU和CPU+NVDLA跑resnet18-cifar10_int8.tmfile，得到的结果虽然都正确，但是结果不一致.......哪的问题待解决,,,,,,,

3to11CHAINJ, 2022-11-12:
> 刚才测试了一下，分别用CPU和CPU+NVDLA跑resnet18-cifar10_int8.tmfile，得到的结果虽然都正确，但是结果不一致.......哪的问题待解决,,,,,,,
[image]
我在zu5ev的板子上部署了nvdla，跑出来的结果是一样的，显示内存double free or corruption。
请问有头绪或者得到解决吗？