OAID/Tengine #1296: yolov5s example failed

chenhu66, 2022-03-06:
I ran your example with the tmfile from your zoo, and got the following error:

./build/install/bin/tm_yolov5s -m models/yolov5s.tmfile  -i pics/ssd_dog_640.jpg
tengine-lite library version: 1.5-dev
Error: input elem num(408000) != reshaped elem num(102000)
Tengine FATAL: Infer node(id: 276, op: Reshape) shape failed.
Tengine: Infer shape of graph failed(-1).
Prerun multithread graph failed.

Coud you please help? Thank you.
[image]

chenhu66, 2022-03-14:
It worked out after changing #define YOLOV5_VERSION6 1 to#define YOLOV5_VERSION6 0

closed 2022-03-14 by chenhu66

Coder-TieHH, 2023-04-11:
Where is the "YOLOV5_VERSION6 1" code

ccc120660, 2023-06-07:
同问，请问找到这代码在哪了吗