OAID/Tengine #1301: Segmentation fault (core dumped) of Tengine Convert Tool !

2050airobert, 2022-03-11:
Hi，
1、 Following the info of [https://docs.khadas.com/linux/vim3/TengineSDK.html，](https://docs.khadas.com/linux/vim3/TengineSDK.html%EF%BC%8C) when I do this “./convert_tool -f darknet -m ~/yolov3.weights -p ~/yolov3.cfg -o yolov3.tmfile” then crash with the error info as below.

---- Tengine Convert Tool ----

Version : v1.0, 15:43:59 Jun 24 2021
Status : float32
Segmentation fault (core dumped)

2、 Then I tried the official open source of OPENAL Tengine ，git clone and build the sdk of the Tengine  repo with the  TIM-VX included. After "make install", in the directory "./build/install/bin"

(1)./convert_tool -h

[Convert Tools Info]: optional arguments:
        -h    help            show this help message and exit
        -f    input type      path to input float32 tmfile
        -p    input structure path to the network structure of input model(*.param, *.prototxt, *.symbol, *.cfg, *.pdmodel)
        -m    input params    path to the network params of input model(*.bin, *.caffemodel, *.params, *.weight, *.pb, *.onnx, *.tflite, *.pdiparams)
        -o    output model    path to output fp32 tmfile

[Convert Tools Info]: example arguments:
        ./convert_tool -f onnx -m ./mobilenet.onnx -o ./mobilenet.tmfile
        ./convert_tool -f caffe -p ./mobilenet.prototxt -m ./mobilenet.caffemodel -o ./mobilenet.tmfile
        ./convert_tool -f mxnet -p ./mobilenet.params -m ./mobilenet.json -o ./mobilenet.tmfile
        ./convert_tool -f darknet -p ./yolov3.weights -m ./yolov3.cfg -o yolov3.tmfile
(2) ./convert_tool -f darknet -p ./yolov3.weights -m ./yolov3.cfg -o yolov3.tmfile
---- Tengine Convert Tool ----

Version     : v1.0, 11:42:03 Mar 10 2022
Status      : float32

----------darknet2tengine begin----------

As you can see above , the error info is almost the same with step 1.
My platform is ubuntu 20.04 server. Anything wrong ?

BR

twmht, 2022-03-15:
solved?

2050airobert, 2022-03-30:
Anyone could help？

2050airobert, 2022-03-30:
Which version of converted tool is the best  till now and there is not so many possible bug？

BUG1989, 2022-04-01:
please try the latest version?

closed 2022-04-01 by BUG1989