OAID/Tengine #1334: { Range ConstantOfShape TopK Not NonMaxSuppression ScatterND } Convert model failed.

vansin, 2022-05-04:
how to new a op for onnx, isn't any example?

```shell
(base) ➜  ~ /project/Tengine/build/install/bin/convert_tool -f onnx -m /media/elaine/data/1.onnx -o /home/elaine/Documents/yolo.tmfile

---- Tengine Convert Tool ---- 

Version     : v1.0, 10:59:41 May  4 2022
Status      : float32

----------onnx2tengine begin----------
Model op set is: 11
Load a Constant node "721" as input[1] of node "Mul_201".
Load a Constant node "730" as input[1] of node "Mul_206".
Load a Constant node "767" as input[0] of node "Add_240".
Load a Constant node "778" as input[1] of node "Mul_247".
Load a Constant node "787" as input[1] of node "Mul_252".
Load a Constant node "824" as input[0] of node "Add_286".
Load a Constant node "835" as input[1] of node "Mul_293".
Load a Constant node "844" as input[1] of node "Mul_298".
Load a Constant node "881" as input[0] of node "Add_332".
Load a Constant node "924" as input[1] of node "Mul_371".
Load a Constant node "931" as input[1] of node "Sub_378".
Load a Constant node "933" as input[1] of node "Mul_380".
Load a Constant node "947" as input[1] of node "Mul_394".
Load a Constant node "1000" as input[0] of node "Less_443".
Load a Constant node "885" as input[1] of node "Where_444".
Load a Constant node "1008" as input[0] of node "Range_451".
Load a Constant node "1090" as input[1] of node "Mul_518".
Load a Constant node "1097" as input[1] of node "Sub_525".
Load a Constant node "1099" as input[1] of node "Mul_527".
Load a Constant node "1113" as input[1] of node "Mul_541".
Load a Constant node "1166" as input[0] of node "Less_590".
Load a Constant node "885" as input[1] of node "Where_591".
Load a Constant node "1174" as input[0] of node "Range_598".
Load a Constant node "1256" as input[1] of node "Mul_665".
Load a Constant node "1263" as input[1] of node "Sub_672".
Load a Constant node "1265" as input[1] of node "Mul_674".
Load a Constant node "1279" as input[1] of node "Mul_688".
Load a Constant node "1332" as input[0] of node "Less_737".
Load a Constant node "885" as input[1] of node "Where_738".
Load a Constant node "1340" as input[0] of node "Range_745".
Load a Constant node "1387" as input[1] of node "Less_781".
Load a Constant node "1392" as input[1] of node "Greater_786".
Load a Constant node "1400" as input[2] of node "NonMaxSuppression_806".
Load a Constant node "1421" as input[0] of node "Range_816".
Load a Constant node "1478" as input[0] of node "Range_862".
Load a Constant node "1487" as input[1] of node "Mul_870".
Load a Constant node "1495" as input[1] of node "Mul_878".
Load a Constant node "1531" as input[0] of node "Less_906".
Load a Constant node "1529" as input[1] of node "Where_907".
Load a Constant node "1539" as input[0] of node "Range_914".
These 6 op are not supported
{ Range ConstantOfShape TopK Not NonMaxSuppression ScatterND }
Convert model failed.

```

wangyaxin1998, 2022-05-19:
解决了吗

tian0zhi, 2023-06-12:
ConstantOfShape 一般是因为原模型（torch）转onnx使用了squeeze函数减少维度，在原模型中用torch.max相应维度代替torch.squeeze使用；topK，Tengine有topKV2不兼容topK，可以使用多次torch.max并将每次取出的位置元素置为小值解决，其他op不清楚解决方案，一个思路就是尽量用兼容算子重写模型。