OAID/Tengine #1375: conversion fail and inference crash

liamsun2019, 2022-08-30:
遇到两个问题：
1. 附档模型用prebuilt的convert tool做转换，会segmentation fault
tm_convert_tool -f onnx -m test.onnx -o test.tmfile
2. 用onnxsim做简化后再转换：
python -m onnxsim test.onnx simplified.onnx
tm_convert_tool -f onnx -m simplified.onnx -o test.tmfile
可以成功，但是推理会crash:
Tengine work node not support 1
Tengine: Device(CPU) failed to init exec node for node(id: 2, type: Convolution, name: 102).
Tengine Fatal: Pre-run subgraph(0) on CPU failed.
Tengine: Scheduler(sync) prerun failed.
Prerun multithread graph failed.

onnx模型本身是工作正常的，转成ncnn和mnn也都可以正常运作。请看一下这个问题，谢谢

[test.zip](https://github.com/OAID/Tengine/files/9452289/test.zip)

liamsun2019, 2022-08-30:
另外一个比较奇怪的现象是，转成tmfile后，relu操作都看不到了

liamsun2019, 2022-08-31:
我用源码编译出convert tool，之前没有经过onnxsim处理的onnx模型可以直接进行转换了，但运行时仍然有相同错误

liamsun2019, 2022-09-01:
inference失败的原因，看起来是因为设定了precision为TENGINE_MODE_FP16的缘故，而转换后的tmfile模型，权重看起来是float32。改为TENGINE_MODE_FP32就可以正常推理。这里再问一下，有没有方法可以在转换tmfile的时候，就把权重转成fp16？MNN和NCNN都有类似的功能。

H2OD, 2023-02-13:
您好，我出现了和您相同的问题，可否留一个联系方式请教一下