OAID/Tengine #1385: 关于yolov7-pose onnx模型转换的问题

lovehuanhuan, 2022-10-08:
onnx convert-tools转换出错。。。
[image]
原始onnx模型地址
链接：https://pan.baidu.com/s/1ALO2cYkoaa4ljfhM8zDbJA 
提取码：vbzp 
--来自百度网盘超级会员V5的分享

Bobe-Wang, 2022-10-12:
这个是slice算子没有处理好，导致的。
diff --git a/source/device/cpu/op/slice/slice_ref.c b/source/device/cpu/op/slice/slice_ref.c
index 037c413b..e619cbc0 100644
--- a/source/device/cpu/op/slice/slice_ref.c
+++ b/source/device/cpu/op/slice/slice_ref.c
@@ -245,10 +245,10 @@ static int onnx_run(const int8_t* in_data, int8_t** out_data, int element_size,
         int start_1 = (param->axis == 1) ? param->begin : 0;
         int start_2 = (param->axis == 2) ? param->begin : 0;
         int start_3 = (param->axis == 3) ? param->begin : 0;
-        int stop_0 = (param->axis == 0) ? (param->end > 0 ? param->end : param->in_shape[0] + param->end) : param->in_shape[0];
-        int stop_1 = (param->axis == 1) ? (param->end > 0 ? param->end : param->in_shape[1] + param->end) : param->in_shape[1];
-        int stop_2 = (param->axis == 2) ? (param->end > 0 ? param->end : param->in_shape[2] + param->end) : param->in_shape[2];
-        int stop_3 = (param->axis == 3) ? (param->end > 0 ? param->end : param->in_shape[3] + param->end) : param->in_shape[3];
+        int stop_0 = (param->axis == 0) ? (param->end > 0 ? param->end : param->in_shape[0]) : param->in_shape[0];
+        int stop_1 = (param->axis == 1) ? (param->end > 0 ? param->end : param->in_shape[1]) : param->in_shape[1];
+        int stop_2 = (param->axis == 2) ? (param->end > 0 ? param->end : param->in_shape[2]) : param->in_shape[2];
+        int stop_3 = (param->axis == 3) ? (param->end > 0 ? param->end : param->in_shape[3]) : param->in_shape[3];
 
         if (param->step > 1)
         {
@@ -301,9 +301,9 @@ static int onnx_run(const int8_t* in_data, int8_t** out_data, int element_size,
         int start_2 = (param->axis == 2) ? param->begin : 0;
 
         // int start_3=(param->axis==3)? param->begin:0;
-        int stop_0 = (param->axis == 0) ? (param->end > 0 ? param->end : param->in_shape_3[0] + param->end) : param->in_shape_3[0];
-        int stop_1 = (param->axis == 1) ? (param->end > 0 ? param->end : param->in_shape_3[1] + param->end) : param->in_shape_3[1];
-        int stop_2 = (param->axis == 2) ? (param->end > 0 ? param->end : param->in_shape_3[2] + param->end) : param->in_shape_3[2];
+        int stop_0 = (param->axis == 0) ? (param->end > 0 ? param->end : param->in_shape_3[0]) : param->in_shape_3[0];
+        int stop_1 = (param->axis == 1) ? (param->end > 0 ? param->end : param->in_shape_3[1]) : param->in_shape_3[1];
+        int stop_2 = (param->axis == 2) ? (param->end > 0 ? param->end : param->in_shape_3[2]) : param->in_shape_3[2];
         // int stop_3=(param->axis==3)? param->end:param->in_shape[3];
 
         for (int n = start_0; n < stop_0; ++n)
@@ -327,8 +327,8 @@ static int onnx_run(const int8_t* in_data, int8_t** out_data, int element_size,
         int start_1 = (param->axis == 1) ? param->begin : 0;
 
         // int start_3=(param->axis==3)? param->begin:0;
-        int stop_0 = (param->axis == 0) ? (param->end > 0 ? param->end : param->in_shape_2[0] + param->end) : param->in_shape_2[0];
-        int stop_1 = (param->axis == 1) ? (param->end > 0 ? param->end : param->in_shape_2[1] + param->end) : param->in_shape_2[1];
+        int stop_0 = (param->axis == 0) ? (param->end > 0 ? param->end : param->in_shape_2[0]) : param->in_shape_2[0];
+        int stop_1 = (param->axis == 1) ? (param->end > 0 ? param->end : param->in_shape_2[1]) : param->in_shape_2[1];
         // int stop_3=(param->axis==3)? param->end:param->in_shape[3];
 
         for (int n = start_0; n < stop_0; ++n)
@@ -507,6 +507,39 @@ static int run(struct node_ops* node_ops, struct exec_node* exec_node, struct ex
         free(input_fp32);
         free(output_fp32);
     }
+    else if (input_tensor->data_type == TENGINE_DT_INT8)
+    {
+        /* dequant to fp32 */
+        int8_t* input_int8 = (int8_t*)input_tensor->data;
+        int8_t* output_int8 = (int8_t*)output_tensor->data;
+        float input_scale = input_tensor->scale;
+        float output_scale = output_tensor->scale;
+
+        float* input_fp32 = (float*)sys_malloc(input_tensor->elem_num * sizeof(float));
+        float* output_fp32 = (float*)sys_malloc(output_tensor->elem_num * sizeof(float));
+        out_data_ptrs[0] = (int8_t*)output_fp32;
+
+        for (int i = 0; i < input_tensor->elem_num; i++)
+        {
+            input_fp32[i] = (float)input_int8[i] * input_scale;
+        }
+
+        ret = ref_slice_common((int8_t*)input_fp32, out_data_ptrs, sizeof(float), &op_param);
+
+        /* quant to int8 */
+        for (int i = 0; i < output_tensor->elem_num; i++)
+        {
+            int udata = round(output_fp32[i] / output_scale);
+            if (udata > 127)
+                udata = 127;
+            else if (udata < -127)
+                udata = -127;
+            output_int8[i] = (int8_t)udata;
+        }
+
+        free(input_fp32);
+        free(output_fp32);
+    }
 
     sys_free(out_data_ptrs);
     if (ret < 0)
diff --git a/source/operator/prototype/slice.c b/source/operator/prototype/slice.c
index 01836909..55f7688f 100644
--- a/source/operator/prototype/slice.c
+++ b/source/operator/prototype/slice.c
@@ -138,7 +138,7 @@ static int infer_shape(ir_node_t* node)
                 }
                 else
                 {
-                    out_dims[i] = dims_in[i] + (slice_end - slice_param->begin);
+                    out_dims[i] = dims_in[i] - slice_param->begin;
                     if (slice_param->step > 1)
                     {
                         out_dims[i] = (out_dims[i] - 1) / slice_param->step + 1;
同时Netron的slice算子显示也是有一些问题的。step并没有在Netron中显示出来。需要修改Netron的代码。

Bobe-Wang, 2022-10-12:
这个是由于Tengine Lite的slice算子没有处理好end = -1时，出现的问题。我之前也遇到过这个问题。见下面的链接slice.txt文件。
[slice.txt](https://github.com/OAID/Tengine/files/9761565/slice.txt)
同时Netron上查看slice算子属性时是有问题的，slice算子中的step并没有显示出来，需要修改Netron部分的代码，重新编译。

lovehuanhuan, 2022-10-27:
仿照yolov5 把focus干掉了

closed 2022-10-27 by lovehuanhuan