OAID/Tengine #1389: 设置armv8.2a后编译报错

niusmall, 2022-10-28:
修改CMakeLists，把TENGINE_ARCH_ARM_82设置为ON
OPTION (TENGINE_ARCH_ARM_82                 "Build ARM v8.2 for ARM platform"           ON)
cmake .. 
make后报错：
1、很多的报错（几百个）：
/home/test/Tengine_v8.2a/source/device/cpu/op/conv/cortex-a/armv8.2/dw_k3s1p1_fp16_a76.S:112: Error: selected processor does not support `fmul v18.8h,v20.8h,v0.h[0]'
2、找不到头文件：
/home/test/Tengine_v8.2a/source/device/cpu/op/conv/cortex-a/armv8.2/conv_dw_kernel_fp16_arm82.c:28:10: fatal error: compiler_fp16.h: No such file or directory
   28 | #include "compiler_fp16.h"
3、找不到头文件：
/home/test/Tengine_v8.2a/source/device/cpu/op/conv/cortex-a/armv8.2/conv_kernel_fp16_arm82.c:30:10: fatal error: conv_kernel_arm.h: No such file or directory
   30 | #include "conv_kernel_arm.h"

请问这些问题该怎么处理？

DamonsJ, 2022-12-19:
同样遇到这个问题
这个是因为什么原因？

code7, 2023-03-22:
same problem. I guess mabey need some ARM's head file.