OAID/Tengine #1396: Matrix multiplication bug

longnp91, 2022-12-09:
I run tmfile inference with Mali. Here is the bug log. Please help!!

lib_path= ['/usr/local/lib/python3.8/dist-packages/pytengine-0.9.1-py3.8.egg/tengine/libtengine-lite.so']
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '5'.
device name:Mali-LODX r0p0 device_vendor:ARM 
self.dtype.enum: 0
self.dtype.enum: 0
self.dtype.enum: 0
self.dtype.enum: 0
self.dtype.enum: 0
unsupported dim num: 2
CPU: failed to run node 616, Mul_265
run subgraph 0 error!
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.26.12) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Traceback (most recent call last):
  File "tengine_conversion.py", line 89, in <module>
    graph.run(1)
  File "/usr/local/lib/python3.8/dist-packages/pytengine-0.9.1-py3.8.egg/tengine/graph.py", line 350, in run
    check_call(_LIB.run_graph(ctypes.c_void_p(self.graph), block))
  File "/usr/local/lib/python3.8/dist-packages/pytengine-0.9.1-py3.8.egg/tengine/base.py", line 110, in check_call
    raise TytengineError(_LIB.get_tengine_errno())
tengine.base.TytengineError: -1

maligawork, 2023-02-13:
@longnp91 Did you fix this problem?

longnp91, 2023-02-14:
> @longnp91 Did you fix this problem?

I haven't fixed it yet.