OAID/Tengine #1400: 文档说明好像有问题，找不到tm_convert_tool

nosleeve1, 2022-12-17:
我在[https://github.com/OAID/Tengine-Convert-Tools](url)
找到了convert_tool执行文件，但是这个文件转onnx模型到tmfile，可视化之后直接维度就不对了，跑的时候一直报infer shape error,不知道是不是convert_tool的原因,这个tm_convert_tool和convert_tool是不同版本的吗

nosleeve1, 2022-12-17:
看了一下基本上Transpose之后tmfile 和 onnx就对不上了

HouXueqing, 2022-12-28:
我的是同样的操作，没有问题，tm_convert_tool可用，onnxsim之后再试下？我的是pth转onnx再onnxsim再转成tmfile，不经过onnxsim会报错，然后他还有个网页版的转换工具也可以试下，虽然我的失败了。。

HouXueqing, 2023-02-20:
Tengine-Convert-Tools/build/install/bin/convert_tool可以找到吗，他应该是更新过版本

   
请问你解决这个问题了吗

imjking, 2023-02-22:
> 

嗯，找到了

HouXueqing, 2023-02-22:
我感觉tengine有点坑，转用paddlelite了。tengine的qq交流群加不进去，github提问甚至知乎找开发者提问都没人理，然后我想在rk3399上用gpu，一直没成功，就放弃了。。。当然他们cpu速度好像还可以，只用cpu问题应该不大

imjking, 2023-02-22:
> 我感觉tengine有点坑，转用paddlelite了。tengine的qq交流群加不进去，github提问甚至知乎找开发者提问都没人理，然后我想在rk3399上用gpu，一直没成功，就放弃了。。。当然他们cpu速度好像还可以，只用cpu问题应该不大
> […](#)

嗯，我现在是onnx转tflite可以用，但是变慢了。tflite量化的时候activation一直报溢出，不知道是不是mean和scale没设对，看源码量化好像没走完整，不知道咋整了。

HouXueqing, 2023-02-22:
我量化时候报过错，后来改的onnx（可能会有版本问题，以及可能需要onnxsim）就好了，你遇到的错误我倒是没见过，或许可以换个激活函数，不看效果，只看能不能跑通，排查一下

imjking, 2023-02-22:
> 我量化时候报过错，后来改的onnx（可能会有版本问题，以及可能需要onnxsim）就好了，你遇到的错误我倒是没见过，或许可以换个激活函数，不看效果，只看能不能跑通，排查一下
> […](#)

嗯