OAID/Tengine #1407: 在RISC-V架构的32位芯片上部署tengine模型

hangxunZ, 2023-02-10:
公司的芯片是基于RISC-V架构的32位MCU芯片，
运行系统：FreeRTOS
请问该怎么移植部署？
尝试将tengine-lite源码编译成静态库.a，然后链接运行。但是无法正常运行，有很多问题，例如：
（1）模型该怎么加载，介绍支持从文件加载，或从memory直接加载。但是没有详细的文档说明，两种方式该怎么配置，特别是从memory加载的方案。麻烦详细说明。
（2）FreeRTOS系统的移植部署流程。文档上讲的全是linux系统的，没有说明FreeRTOS移植适配的流程。
（3）平台有移植适配32位的RISC-V架构运行FreeRTOS系统的MCU芯片吗？是否能够正常运行？

谢谢！

ccc120660, 2023-04-23:
同想请问这个问题，请问有答案了吗

ComingToy, 2023-05-08:
请问具体是什么芯片？我来尝试一下能不能移植过去

ccc120660, 2023-05-15:
> 请问具体是什么芯片？我来尝试一下能不能移植过去

大佬考虑试试c906嘛

ComingToy, 2023-05-15:
> > 请问具体是什么芯片？我来尝试一下能不能移植过去
> 
> 大佬考虑试试c906嘛
我现在手上刚好有一块运行Linux的c906，不过楼上似乎想要的是freertos的riscv之类的mcu环境

ComingToy, 2023-06-25:
> > 请问具体是什么芯片？我来尝试一下能不能移植过去
> 
> 大佬考虑试试c906嘛

最新的C906适配在这里[ComingToy/Tengine](https://github.com/ComingToy/Tengine)

ccc120660, 2023-06-25:
> 

大佬让我泪目，我现在还在整硬件，整完了我再来试试您改的这个适配，谢谢大佬

ccc120660, 2023-06-25:
/
/

ccc120660, 2023-06-25:
> > > 请问具体是什么芯片？我来尝试一下能不能移植过去
> > 
> > 
> > 大佬考虑试试c906嘛
> 
> 最新的C906适配在这里[ComingToy/Tengine](https://github.com/ComingToy/Tengine)

.

ccc120660, 2023-06-25:
> 最新的C906适配在这里[ComingToy/Tengine](https://github.com/ComingToy/Tengine)

大佬让我泪目，我现在还在整硬件，整完了再来试试您的这个版本，谢谢大佬

ccc120660, 2023-09-12:
> 
您好，冒昧打扰，能请问下您是在什么开发板测试的呢，是全志D1的那个板子嘛= =

ccc120660, 2023-09-12:
> > > 请问具体是什么芯片？我来尝试一下能不能移植过去
> > 
> > 
> > 大佬考虑试试c906嘛
> 
> 最新的C906适配在这里[ComingToy/Tengine](https://github.com/ComingToy/Tengine)

您好，冒昧打扰，能请问下您是在什么开发板测试的呢，是全志D1的那个板子嘛= =

ComingToy, 2023-10-06:
> > > > 请问具体是什么芯片？我来尝试一下能不能移植过去
> > > 
> > > 
> > > 大佬考虑试试c906嘛
> > 
> > 
> > 最新的C906适配在这里[ComingToy/Tengine](https://github.com/ComingToy/Tengine)
> 
> 您好，冒昧打扰，能请问下您是在什么开发板测试的呢，是全志D1的那个板子嘛= =

哪吒开发板上测试的，是全志D1