OAID/Tengine #1413: yolov5 (no focus) int8 quant后，在嵌入式平台上跑到run_graph函数就挂了

zhangliming0113, 2023-04-06:
1.平台信息：
   rv1126
   linux

2. log信息：
tengine-lite library version: 1.5-dev
mean:(123.675003,116.279999,103.529999) scale:(0.017125,0.017507,0.017429) input_scale:0.034134 zero_point:0 
Input data type 2 not to be supported.
CPU: failed to run node 128, Sigmoid_1
run subgraph 0 error!
Run graph failed

3. 不管跑npu还是cpu，现象一样

zhangliming0113, 2023-04-07:
增加hardswish_kernel_ref_int8即可解决，，mark一下

closed 2023-04-07 by zhangliming0113

ccc120660, 2023-06-05:
请问大佬这个文件里面的内容是需要自己写吗？

ccc120660, 2023-06-05:
[image]
[image]
R)1SBR](https://github.com/OAID/Tengine/assets/64881480/41c0d5aa-ce7f-4d7b-a1e4-035d3777dea7)
[image]

请问大佬这样修改就可以了嘛，我修改了之后删除build文件夹重新编译，还是会有报错。。。

zhangliming0113, 2023-06-05:
这个文件修改没有问题，，确认调用到了嘛？在hardswish_ref.c里run函数增加
else if (input_tensor->data_type == TENGINE_DT_INT8)
        ret = ref_hardswish_int8(input_tensor, output_tensor);

ccc120660, 2023-06-05:
非常非常感谢大佬，我没有修改您提到的run函数，现在已经好使了，非常非常感谢

ccc120660, 2023-06-06:
不知道能不能厚着脸皮请教下大佬，量化时输入的参数:mean和scale是每个数据集都不一样吗，我看网上的coco数据集用的都是mean为0和scale为0.003921，我自己的数据集是voc格式的舰船的数据集，在训练前没有自己进行预处理（我看yolov5自己有预处理操作）（之前用的yolox，各种参数都试了但是效果很差，所以换成yolov5s试试），我看issue里面大家的参数好像都是自己定义的，所以不知道能不能请问下大佬该如何求取这个参数。冒昧打扰非常抱歉= =

zhangliming0113, 2023-06-07:
mean和scale应该是一样的，其实就是归一化而已

ccc120660, 2023-06-07:
谢谢您的回答（我把hardswish换成relu，未量化之前用tm_yolov5s可以推理，量化之后用tm_yolov5s_int8推理就显示Segmentation fault (core dumped)，看来是g了），不管怎样，感谢您的回答，祝您工作顺利